Ultimaker/Cura #16704: Can't Choose how long layers thick I'd like the draft shield for multi-colored prints

joseuzair718, 2023-09-09:
### Is your feature request related to a problem?

Trying to increase the number of walls for the draft shield. Sometimes the IDEX printers like my Sovol 4 needs a draft shield to "prime" the nozzle so there's no dots or pots, an in prusa you can select how many layer high and how many layer thick you want the draft shield.

I searched online, someone mention change the nozzle size to 0.41 and it'll give you a 2 wall draft shield, but that didn't work for me.

Anyone have any ideas or is this not a feature in Cura atm?

Any help would be greatly appreciated, as I switch back and forth between Cura and Prusa for certain features but Cura is my preferred slicer and this is one of the main features that makes me turn to Prusa since I got an IDEX, I tend to use it a lot and draft shields are great.

I know even with a single wall draft shield it should print find as long as my XY offset is set perfectly on my Sovol but I find when doing PLA and TPU prints, the PLA tends to slip off and I end up having to add painters tape around the draft shield to let the next layer of TPU stick down.

### Describe the solution you'd like

An option for the number of walls for the draft shield will allow draft shields that are more reliable during prints, especially when printing TPU and PLA on IDEX printers.
[image]

### Describe alternatives you've considered

Prusa Slicer
[image]

### Affected users and/or printers

Anyone with an IDEX printer, ESPECIALLY when printing TPU and PLA. Also Anyone with an IDEX in general since XY calibration isn't always perfect so having a draft shield 2 - 4 walls wide will avoid a failed print. My prints kept failing with draft shields until I switched to Prusa Slicer then the draft shield never had issues after that. However there are some features in CURA where I had no choice but to use a single wall draft shield, however I was lucky it was only a 50mm tall print so it wasn't a big deal, I only have to place blue tape 4 - 5 times when the PLA decided not to bond with the TPU below it. However when printing the same print as a test with Prusa with 3 walled draft shields, worked great. When I searched online, others had this issue and also wish Cura had this feature but they all said they switched to Prusa. I use both Prusa and Cura, but I'm much more familiar with Cura and would love to stick with it.

### Additional information & file uploads

[image]

GregValiant, 2023-09-09:
That is a well written and well thought out argument.
Cura doesn't natively support IDEX printers (although some can be made to work quite well).
Ultimaker doesn't manufacture any IDEX printers and I seriously doubt they will pay their people to work on providing software for another companies printer.  So I would guess that if this is going to happen it will have to come from someone in the community.

joseuzair718, 2023-09-10:
Man that s*cks... All they have to decide is how many walls are on the draft, exactly the same way with the skirt...

Oh well, PSlicer it is then, until they fix this...,.

MariMakes, 2023-09-11:
Hey @joseuzair718,

Thanks for your nice request👍
@GregValiant is right, I've never worked with IDEX printers. 
But I have worked with Dual extrusion printers. 

There we use Prime Towers to prime the nozzles.
There is a number of settings that you can use to increase the amount of filament purged. 
[image]

Can you help me understand why prime towers would not work for you? 
Maybe you can combine it with an Ooze Shield?