Ultimaker/Cura #16706: AArch64 version not running/available

ClaytonRehder, 2023-09-10:
### Cura Version

5.4.0

### Operating System

ChromeOS (Debian 11)

### Printer

Ender-3 v2

### Reproduction steps

Try to install the .appimage on an AArch64 Linux System (ChomeOS- Debian 11)

### Actual results

Can not read the binary

### Expected results

Run the application

### Add your .zip and screenshots here ⬇️

Not needed/do not have

smartavionics, 2023-09-11:
Hi, you could try my Cura variant as I provide an aarch64 appimage built for raspberry pi os. Unfortunately, it's based on Cura 4 not 5.

https://github.com/smartavionics/Cura/releases

castlec, 2023-12-21:
> Hi, you could try my Cura variant as I provide an aarch64 appimage built for raspberry pi os. Unfortunately, it's based on Cura 4 not 5.
> 
> https://github.com/smartavionics/Cura/releases

@smartavionics I have not dug into what your build adds to the 4 series so I don't know if this information helps you much. I have put a bunch of effort into getting the 5 series running on aarch64. My notes are [here](https://github.com/Ultimaker/Cura/issues/17677#issuecomment-1865456981).

@jellespijker maybe mark 17677 as a duplicate of this issue? I didn't do a good job of searching before opening, it seems.