Ultimaker/Cura #16707: wood part v5

astromanED, 2023-09-10:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

ender 3 pro

### Name abnormal settings

none

### Describe model location

center

### Describe your model

hexagonal cup with deep grooves

### Add your .zip here ⬇️

[CE3PRO_wood part V5 stl.zip](https://github.com/Ultimaker/Cura/files/12568387/CE3PRO_wood.part.V5.stl.zip)

I had a V4 that works however by expanding the gap in the side the slicing has failed. I have attempted many different settings to no avail

GregValiant, 2023-09-10:
Thanks for the report and the file.
There is an oft-reported bug in 5.4.0 which can keep good models from slicing.
There are three workarounds that can often fool the bug.

- Move the model away from the Cura build plate midpoint.
- Rotate the model in Cura
- Change the Initial Layer Height by a little bit.

In this case I had to do all three.  This is rotated, moved, and the Initial Layer Height is 0.25.
I'll mark this as a duplicate slicing error.  This model is a vertical version of some buildings that have wood or stone floors that cause the same sort of problem.  I'm going to leave it open and maybe the Cura team can use this one for debugging.
[image]

astromanED, 2023-09-16:
sorry for the late reply. Yes! this worked. next step is to keep the detail needed. Im using CHEPs EF settings. again, perfect on my V4 but some fidelity lost on the V5.

Thanks for the pointers.

> Thanks for the report and the file. There is an oft-reported bug in 5.4.0 which can keep good models from slicing. There are three workarounds that can often fool the bug.
>
> * Move the model away [...]

closed 2024-04-03 by HellAholic