Ultimaker/Cura #16714: Slicing defects

kevinjones4642, 2023-09-11:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

S5

### Reproduction steps

If I slice the same part with the same settings in 5.3.1 and 5.4.0, the part sliced in 5.4.0 has occasional defects at random layers, where it seems to drop points in the path, yielding jagged defects. 

### Actual results

Areas that are smooth in CAD, and smooth when sliced in Cura 5.3.1 are jagged when sliced in Cura 5.4.0

### Expected results

I would not expect the above behavior

### Add your .zip and screenshots here ⬇️

[STL.zip](https://github.com/Ultimaker/Cura/files/12578985/STL.zip)

GregValiant, 2023-09-12:
Thanks for the report.
The imperfections I see in the print I believe have been reported previously.  Unfortunately I can't remember the title of the report.  I do remember seeing those imperfections and I investigated and found that they do show up in the gcode.

Someone from the Cura team will take a look.

kevinjones4642, 2023-09-12:
Thanks for the feedback. I was too lazy to search for similar posts. I
should have guessed that it was already in there. I rolled back to 5.3.1
for now.

fieldOfView, 2023-09-12:
Try lowering the `Maximum Resolution` setting (under `Mesh Fixes`) to 0.2. This has fixed similar reports, but I don't quite understand why it would be necessary in newer versions of Cura.

kevinjones4642, 2023-09-12:
Thanks - that fixes it. There are so many settings that I've never looked
at.

--Kevin

kevinjones4642, 2023-09-12:
Actually, 0.2 fixed most of the problems, but a few spots still have some
small defects. 0.1 seems to clean up those.