Ultimaker/Cura #16715: Multiple extruder Error with BIBO 2 dual

nilsbeet, 2023-09-11:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Standard BIBO 2 dual extruder

### Reproduction steps

slcing a stl file, selected to print with extruder 2

### Actual results

Extruder 1 heats up and starts to print.

Problem uccurs only when using v5.4.0 previously been using v4.12.1 and it´s all been working fine

### Expected results

printing with ex2

### Add your .zip and screenshots here ⬇️

[GCODE of issue.zip](https://github.com/Ultimaker/Cura/files/12579013/GCODE.of.issue.zip)

GregValiant, 2023-09-11:
Thanks for the report and the file.
This does look like a bug.
Cura is correctly calling for T1 prior to your Startup Gcode.  Then, your StartUp Gcode ends with "T{initial_extruder_nr}" as you would want.
The problem seems to be that although Cura called out the correct extruder initially, the variable {initial_extruder_nr} has the wrong value and so that last line of your StartUp is enabling T0 instead of T1.
I think what happened is that although there is no support in the gcode that the "Support Extruder" (which is Extruder 1) is still being assigned and initial layer support always prints right after the skirt/brim.

There seems to be a workaround and that is to explicitly set the "Support Extruder" to Extruder 2 if that is the one you want to start the print.  In this case you would need to enable "Generate Support" then switch the support extruder to extruder 2, and then turn supports off again.

@MariMakes and the Cura team will take a look.

MariMakes, 2023-09-18:
Hi @nilsbeet,

Welcome to the UltiMaker Cura Github 🚀 
Sorry, it took us a while to get back to you 😞 

I've shown your issue to the team, and it's indeed with most dual extrusion printers. We must have missed it because printing with only the second extruder is not a scenario we often cover during testing.

We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-11053

Thanks for the report! 👍