Ultimaker/Cura #16716: Search Through Printer Names

lukasmwerner, 2023-09-11:
### Is your feature request related to a problem?

When instructing new users to use cura it often is difficult to go parse through the 500+ Printer Companies and thousands of printers. 

### Describe the solution you'd like

If there was a search bar for going through all the printers that would significantly simplify adding new printers.

### Describe alternatives you've considered

N/A

### Affected users and/or printers

New users.

MariMakes, 2023-09-15:
Hi @lukasmwerner,

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

We have grown into a significant amount of available printers now, and a feature like this would help us as well. 
We've added a ticket to the backlog with the intent to improve this. 
For internal reference CURA-11003

Thanks for your report 👍
Thanks for the report! 👍

closed 2024-04-22 by saumyaj3

MariMakes, 2024-07-16:
👋 Quick update on our side.
This is resolved in the 5.8 release 🎉 , you can download the Beta version with the fix here: [https://github.com/Ultimaker/Cura/releases/tag/5.8.0-beta.1](https://github.com/Ultimaker/Cura/releases/tag/5.8.0-beta.1)

Thanks again, and please let us know if you run into any other issues 💪