Ultimaker/Cura #16718: 5.5 alpha (nightly build) not slicing because it cannot connect to the CuraEngineGradualFlow plugin

fieldOfView, 2023-09-12:
### Cura Version

5.5.0 alpha 2023-09-11

### Operating System

Windows 10

### Printer

UM

### Name abnormal settings

none

### Describe model location

irrelevant

### Describe your model

irrelevant

### Add your .zip here ⬇️

Relevant log:
```
2023-09-12 11:00:12,624 - INFO - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [339]: Starting to slice...
2023-09-12 11:00:12,625 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend.slice [359]: Going to slice build plate [0]!
2023-09-12 11:00:12,984 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._onStartSliceCompleted [571]: Sending slice message took 0.35899806022644043 seconds
2023-09-12 11:00:13,223 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2023-09-12 11:00:13.230] [info] Create local channel on port 50674.
2023-09-12 11:00:15,281 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2023-09-12 11:00:15.282] [error] failed to connect to all addresses; last error: UNAVAILABLE: ipv4:127.0.0.1:50674: WSA Error
2023-09-12 11:00:16,285 - ERROR - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [1029]: Backend exited abnormally with return code 3221226505!
2023-09-12 11:00:16,292 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [412]: Attempting to kill the engine process
2023-09-12 11:00:16,293 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [418]: Killing engine process
2023-09-12 11:00:16,293 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [421]: Engine process is killed. Received return code 3221226505
2023-09-12 11:00:16,294 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [294]: Previous socket existed. Closing that first.
2023-09-12 11:00:16,305 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [232]: Socket state changed to Listening
2023-09-12 11:00:16,309 - INFO - [MainThread] UM.Backend.Backend.startEngine [96]: Started engine process: C:\Program Files\UltiMaker Cura 5.5.0-alpha+testing_8126b9\CuraEngine.exe
2023-09-12 11:00:16,310 - DEBUG - [MainThread] UM.Backend.Backend._backendLog [140]: [Backend] Calling engine with: ['C:\\Program Files\\UltiMaker Cura 5.5.0-alpha+testing_8126b9\\CuraEngine.exe', 'connect', '127.0.0.1:49676', '']
2023-09-12 11:00:16,348 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2023-09-12 11:00:16.347] [info] Connecting to 127.0.0.1:49676
2023-09-12 11:00:16,349 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [236]: Socket state changed to Connected
2023-09-12 11:00:16,350 - DEBUG - [MainThread] UM.Backend.Backend._onSocketStateChanged [225]: Backend connected on port 49676
2023-09-12 11:00:16,455 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2023-09-12 11:00:16.455] [info] Connected to 127.0.0.1:49676

```

Disabling my firewall did not fix this, disabling the CuraEngineGradualFlow plugin did.

Full log here:
[cura5.5alpha log.zip](https://github.com/Ultimaker/Cura/files/12584692/cura5.5alpha.log.zip)

jellespijker, 2023-09-12:
This is indeed a known issues, the culprit is this:
https://github.com/Ultimaker/CuraEngine_plugin_gradual_flow/blob/main/CuraEngineGradualFlow/GradualFlowPlugin.py#L38

You can change it locally for now to return True.

The problem is that the engine expects that the gradual_flow service is started but we actually don't start itdue to this check. It should be fixed tomorrow or the next day at the latest

jellespijker, 2023-09-12:
Internal ticket: CURA-11031

closed 2023-09-13 by rburema