Ultimaker/Cura #16726: "Print Thin Walls" keeps failing to slice

WilmerSuarez, 2023-09-14:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Creality Ender-3

### Name abnormal settings

Enabled "Print thin Walls". Slices otherwise but not complete.

### Describe model location

If  I rotate 90 degrees, it does.

### Describe your model

Yes, yes, and yes.

### Add your .zip here ⬇️

[CE3E3V2_arabin warrior_fixed.zip](https://github.com/Ultimaker/Cura/files/12605072/CE3E3V2_arabin.warrior_fixed.zip)

GregValiant, 2023-09-14:
Thanks for the report.
I rotated the model 90° CCW in Cura and it sliced.
This is a known bug and will hopefully be fixed in the next release.
I'll mark this as a duplicate report and close it.  The Cura team has lots of examples of this behavior.  I hope you understand.

closed 2023-09-14 by GregValiant

Kiefels2, 2023-11-26:
I have recently ran into this problem AGAIN. Went away for a while but now it's back again, even on models it sliced the day previous.
Tried the new beta 2 version and it does it in that also.
I have not added any new hardware or sofware, so at a loss on why this came back
I can only slice if I invert the model which on most models creates bigger printing problems