Ultimaker/Cura #16727: Lion_wall_lighting

Chris76-3D, 2023-09-14:
### Cura Version

5.4.0

### Operating System

windows 11

### Printer

Wanaho D12 230

### Name abnormal settings

Non

### Describe model location

non

### Describe your model

non , oui ,oui

### Add your .zip here ⬇️

[Great_Lion_wall_cf561a6220.zip](https://github.com/Ultimaker/Cura/files/12606884/Great_Lion_wall_cf561a6220.zip)

GregValiant, 2023-09-15:
Thanks for the report.
This is the third report about that model and once again I can't duplicate the problem as it slices for me.
There is a slicing bug in 5.4.0 that can keep good models from slicing.  Have you tried rotating, or moving the model?  Sometimes changing the Initial Layer Height by just a little (like from 0.20 to 0.23).

I'll mark this as a duplicate and go ahead and close it.  The Cura team has a lot of examples of this behavior.  I hope you understand.

5axes, 2023-12-08:
Duplicate : https://github.com/Ultimaker/Cura/issues/17310