Ultimaker/Cura #16730: slicing with experimental break support into chunks

colemi4117, 2023-09-14:
### Cura Version

5.4

### Operating System

windows 11

### Printer

ender 3 pro

### Name abnormal settings

break up support in chunks

### Describe model location

center

### Describe your model

moose 

### Add your .zip here ⬇️

[CE3PRO_moosetana 036.zip](https://github.com/Ultimaker/Cura/files/12610323/CE3PRO_moosetana.036.zip)

GregValiant, 2023-09-15:
Thanks for the report and the file.
When Cura paints part of a model with a polka-dot pattern it means there are errors in the model.
[image]

The errors need to be fixed or you can get bad slices (when it does slice).

This is the repaired model.  I had to remove the support blockers in order to export the file.
[image]

As you see it's a 50+ hour print.

I'll mark this as a duplicate error and the Cura team will take a look.

colemi4117, 2023-09-15:
Thank you for your time on this.  Please tell me, how did you repair the
model?  Did you do it in Cura or in another application?

closed 2024-04-03 by HellAholic