Ultimaker/Cura #16733: Slicing Failure

RogerPlaysMC, 2023-09-15:
### Cura Version

5.4.0

### Operating System

Window 11

### Printer

Anycubic Kobra Max (actually a Kobra Plus, but option was not available)

### Describe your model

First time slicing this, 4 pieces of Mandalorian armor from thingaverse

### Add your .zip here ⬇️

File is 49.8MB, here is the We Transfer link. https://we.tl/t-uY9vOn7Sfb

GregValiant, 2023-09-15:
Thanks for the report and the files.
They all sliced for me "as received" so I can't duplicate the problem.  There is a slicing bug in 5.4.0 that can keep models from slicing and it seems more prevalent on Windows 11 machines.
There are three workarounds that can often fool the bug.

- Rotate the model on the Cura build plate
- Make a small adjustment to the "Initial Layer Height"
- Move the model to a different place on the Cura build plate.

I'll mark this as a duplicate report and the Cura team will take a look.

RogerPlaysMC, 2023-09-15:
Rotating it did solve the issue.

GregValiant, 2023-09-15:
I have a suggestion...
The tall narrow pieces should be printed with the wide part facing the front of the machine.  You don't want the bed slamming back and forth trying to throw the print onto the floor when it gets tall.
With that in mind you might consider adding an "M201 Y350" line to the gcode at around layer 400.  That will soften the starts and stops of the "Y" axis so the print doesn't get jerked back and forth.

closed 2023-09-15 by GregValiant

RogerPlaysMC, 2023-09-15:
> I have a suggestion... The tall narrow pieces should be printed with the wide part facing the front of the machine. You don't want the bed slamming back and forth trying to throw the print onto the [...]

I appreciate the suggestion! I haven't had that issue yet, as I cover the need in a thin film double sided tape before printing, so I usually need to wrestle the pieces off. I will definitely consider it for future prints.