Ultimaker/Cura #16738: Project fails to slice with some horizontal hole expansion values

benbelly, 2023-09-15:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Ender 3 Pro

### Name abnormal settings

Hole Horizontal Expansion = 0.1. If I change to 0.12 it slices. 0.6 nozzle

### Describe model location

Nope

### Describe your model

Yes

### Add your .zip here ⬇️

As mentioned, it slices with no horizontal expansion, and with a horizontal hole expansion of 0.12. Fails a 0.2, succeeds at 0.22. I'm using a 0.6 nozzle.
[DimmerSwitch.zip](https://github.com/Ultimaker/Cura/files/12618946/DimmerSwitch.zip)

5axes, 2023-09-15:
- Even if the CuraEngine should not crash , your issue is due to the shape of the very small grooves on the inner diameter. This geometry is not printable with this size of nozzle, and it would have been preferable to have a completely smooth bore.
- 
[image]

benbelly, 2023-09-15:
Ha! Thanks. I didn't realize those were there. I'll clean that up

GregValiant, 2023-09-15:
It sliced with an Initial Layer Height of 0.29.  That makes it look to be a duplicate of the normal 5.4.0 slicing bug.
I'll mark this as a duplicate and go ahead and close this as the Cura team has lots of examples of this behavior.  I hope you understand.

As @5axes says, getting rid of those inner features is a good idea since they won't slice anyway.  They probably aren't the cause for slicing to fail as this sort of thing has been reported numerous times.  The slicing bug seems to be more prevalent in Win11 machines.

closed 2023-09-15 by GregValiant