Ultimaker/Cura #16739: Notifications permanente printer connected successfully

alexisbouyon, 2023-09-15:
### Cura Version

5.4

### Operating System

Windows 11

### Printer

Flsun V400

### Reproduction steps

Installation d'une camera sur le pad de la flsun V400

### Actual results

J'ouvre cura pour paramétrer ma pièces et j'ai des notification de connexion en permanence 
Flsun V400 printer connected successfully

### Expected results

Supprimer les notifications ou possibilité de les masquer

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2023-09-15:
Thanks for the report.
Maybe @MariMakes can help with this.  I assume you have the USB connected, are you printing over the USB or is there some connection to the camera?

alexisbouyon, 2023-09-16:
Bonjour 
La connexion se fait par le wifi
La caméra est connectée au pad de l'imprimante en usb

GregValiant, 2023-09-16:
If you are not using USB Printing plugin to print then I would suggest turning it off.
Go to the Marketplace and select the "Gear" icon.  Scroll down and disable "USB Printing".  It keeps trying to connect to a USB device that might be enabled as a serial device but not a printer.

alexisbouyon, 2023-09-16:
J'ai désactivé le plugin et c'est toujours pareil