Ultimaker/Cura #16746: A model mirrored on top of itself fails to slice.

xykobas3rd, 2023-09-16:
### Cura Version

5.4

### Operating System

Windows 10

### Printer

Ultimaker 3

### Name abnormal settings

Does not slice with default settings either

### Describe model location

No

### Describe your model

Two asymetrical pieces merged to one

### Add your .zip here ⬇️

[UM3_spooliev3.zip](https://github.com/Ultimaker/Cura/files/12640148/UM3_spooliev3.zip)

The model is a small asymmetrical piece that has been copied, pasted in the exact same location, and mirrored across a plane in order to create one single symmetrical piece.
It fails to slice only when they have the same location.
By shifting one model -even by 0.1mm away from the other- it slices successfully . But that defeats the point.

GregValiant, 2023-09-18:
Thanks for the report.
There is something about the models when they overlap that Cura finds un-sliceable.
A workaround is to merge the models.  You are on Windows and MS 3D Builder can do that.
I was also able to get it to slice by turning off "Remove Mesh Intersection".  There are a couple of bug reports regarding that setting so this report might be a duplicate.
The Cura team will take a look.

I have found that it's always better to give Cura the model you want rather than do some sort of assembly in Cura.  This slice is with the models "merged" in MS 3D Builder.  There is only a single STL in this slice.
[image]

xykobas3rd, 2023-09-20:
Oh. Interesting. Hadn't tried Slic3r yet. 
It was a simple little model I drew up so I just mirrored the features in CAD.
Was just trying to be lazy by doing it in Cura. 😝

GregValiant, 2023-09-20:
What you are trying to do really is a CAD operation rather than something that you would do in a slicer.
I was able to slice the double model but it came out better with the models merged in 3D Builder.
I'll leave the bug label on this and leave it open so the Cura team can make the call.

Here is the double model.  You can see that Cura wants to respect that there are two models that overlap.  This is with "Remove Mesh Intersection" enabled and "Alternate Mesh Removal" enabled.  Those lines are going to be in the print.
[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic