Ultimaker/Cura #16747: Dual extruder mixer hotend - no purge block nessasary

SynapticSintax, 2023-09-16:
### Is your feature request related to a problem?

Purge blocks and/or retracting and moving forwards the filaments within the hotend are either wasteful or just don't function well due to oozing. 

### Describe the solution you'd like

 If a setting were to permit the user to change the extruder sooner than the way Cura does at the moment for dual hotend users dependent upon a variable defined by the user, then the hotend could swap over extruders before getting to the area to be swapped and extrude what was left in the nozzle while the desired extruder for the uppcoming area is used to clear out the hotend of the previous extruder.  I'm thinking of something like the coasting function but instead of turning off all extruding, switching to the incoming extruder.  

### Describe alternatives you've considered

I have been looking into manipulating the gcode after the slicing through scripting and unfortunately am new to python, and it's capabilities, not to mention the time I have to focus on this is a bit low.  Scripting seems to be something that is feasible but clumsy and unnecessary if I could add a setting, yet I would have to figure out how Cura settings function and where they are within the code, and that is where I am at, just a bit frustrated so I figured I would ask if it would be something that Cura would like to have in it's repertoire, as I cannot guarantee I can do it. 

### Affected users and/or printers

Any printer with a mixer hotend will be able to use this yet for any single extruder or multi extruder/w multiple hotends will not find this setting useful.

GregValiant, 2023-09-17:
There is a post processor "ColorMix".  It supports two extruders.
I decided to start with a clean sheet and write one for both 2-in-1-out and 3-in-1-out hot ends.  This post processor does not function the same as letting Cura do the tool changes.  The post processor does not use a purge tower and it needs to be disabled in Cura.  If "Park and Purge" is selected it just moves to a location and spits out material, retracts, and goes back to printing.

Unzip the file and put "MultiExtColorMix.py" into the "scripts" folder in your Configuration Folder.  I have a Pull Request in on this but your request here meant some changes.  I added an option to the Park and Purge function for "Initial Purge Only".  That will start an extruder out with the mix ratio you want and all other tool changes will just keep going without any stops.  It sounded like that's what you wanted.  
This post processor handles all the tool changes.  You can run multiple instances and change the mix during a print.  It does have a learning curve.
You can post any comments or complaints on the PR #16176 page.  Remember that post processors are "for everyone" and not any particular special situation.  I only have a "virtual" multi-extruder printer for debugging so you might run into something I didn't anticipate.
[MultiExtColorMix.zip](https://github.com/Ultimaker/Cura/files/12643036/MultiExtColorMix.zip)

SynapticSintax, 2023-09-17:
I was more speaking of a switch of 100% T0 to a 100% T1 with a early switch as the like the coast function in Cura, but instead of stopping all extruders a user defined amount dependent upon the size of the user heatblock.  That way when the nozzle comes to the different filament area the new filament is already at the nozzle... No purge, no retract, no stops necessary.  
Are you asking for me to go to PR #16176 page to continue communications?  
I downloaded the zip, and will see how it works code wise.  Mixing isn't my issue.

GregValiant, 2023-09-18:
It's a niche situation.  Most people want a clean break between colors.
I suppose you could do something like figure the distance of filament required to clear the nozzle (given the layer height and line width) and then insert the tool change at a corresponding E value prior to where it is in the gcode.  I can see where blobs or material shortages at the nozzle could be a problem.  At .2 layer height and .4 line width - 1mm of 1.75 filament is about 30mm of extrusion.  Things can get complicated very quickly.

SynapticSintax, 2023-09-18:
Yes, exactly. 
  I'm looking into the Cura functions around the coast setting, and have already figured that this is best suited for a plugin not a script, I just have to learn my way in Cura and making plugins for Cura, both of which have not the most in depth knowledge base for me to study from.  I can code and I can do maths without issue, it will just take me some time to get the job done, and was reaching out to those with knowledge about these systems floating the idea to see if it was of interests.  This is necessary function for a multi in 1 out system to really work correctly, but it is a niche setup and I get that.

closed 2023-09-18 by SynapticSintax