Ultimaker/Cura #16749: Bottom layer thickness not affected on extruder 2

fmgrotepass, 2023-09-17:
### Cura Version

5.4.0

### Operating System

Linux

### Printer

Ultimaker UM3

### Reproduction steps

Choose a model with hollow core and open top and bottom
Choose extruder 2
Enable Spiralize outer contour
Choose bottom thickness 0
[BearingCovers.stl.zip](https://github.com/Ultimaker/Cura/files/12642338/BearingCovers.stl.zip)
[image]
[image]

### Actual results

In the preview it is clear that the first layers has the bottom printed. 

### Expected results

The way it behaves with extruder 1 is what is expected. 

### Add your .zip and screenshots here ⬇️

[BearingCovers.stl.zip](https://github.com/Ultimaker/Cura/files/12642354/BearingCovers.stl.zip)
[image]
[image]

GregValiant, 2023-09-18:
Thanks for the report.
I can make it have a floor but the floor goes away when I have Ext2 set to all zeros in the Top/Bottom Settings.
The Cura team will take a look.

Here is your model set to Extruder#2 and Spiralize is on.
[image]

GregValiant, 2023-09-21:
Did you get this to work?

GregValiant, 2023-09-21:
I'll go ahead and remove the bug label and close this as it appears to be a settings issue.  Let me know if you feel that is premature.

closed 2023-09-21 by GregValiant