Ultimaker/Cura #16752: Gap between wall and bottom with lines pattern depending on location of the part

EmJay276, 2023-09-18:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Ultimaker 2+

### Reproduction steps

1. Add a Cube (20 x 20 x ? mm) at (0, 0, 0)
2. Linewith: forced 0.4mm --> results in exactly 50 lines)
3. Top/Bottom Pattern: Lines
4. Top/Bottom Line Directions: [ 90 ] (to get bottom lines parallel to the walls)

On one side is a gap between the walls (left) and bottom lines, of the other side an overlap (right)
[image]

5. Move Cube center to (0, 0.1, 0)

Gaps / Overlap disappear
[image]

### Actual results

Gaps between the walls and top/bottom **depending on the location of the part**
[image]

### Expected results

**Always** no gaps between the walls and top/bottom
[image]

### Add your .zip and screenshots here ⬇️

[gaps_bottom_wall.zip](https://github.com/Ultimaker/Cura/files/12648106/gaps_bottom_wall.zip)

GregValiant, 2023-09-18:
Thanks for the report.
I see the gap on one side at 0.26mm and the gap on the other at 0.46mm so moving the model .1 will even the gap.
I knew this happened in previous versions with Infill but I was unaware that the skins have the same effect.  

The Cura team will take a look.  I have not seen this reflected in the quality of my prints, but the effect does make it into the gcode.

EmJay276, 2023-09-18:
Thanks for your fast reply!

In prints this results in a noticeable gap with unconnected lines. I don't have a photo with me right now but it definitely affects the structural integrity of the print.

GregValiant, 2023-09-18:
I almost always print with the skins at 45° to the exterior walls and with the infill at 90° to the walls.  As a result I wouldn't notice this as a matter of course.  If the part has to be rotated 45° then I change the line directions of both the skins and the infill to maintain the angles I like.  I level by hand and I always run my Initial Layer Flow at 105% just in case.  I figure a little over-extrusion is better than a little under-extrusion.

EmJay276, 2023-09-18:
For my personal prints I also do this, but I prefer to set the z offset of my nozzle a little "to low" to get some extra squeeze on the first layer. So I can keep the flow constant for all parts of the print.

For structural parts I print at work, the line direction greatly affects the performance of the finished parts, especially with filled filaments.