Ultimaker/Cura #16753: Slicing Failed.

frosty5350, 2023-09-18:
### Cura Version

5.4.0

### Operating System

Windows 10 Pro

### Printer

Crealty Ender 3

### Describe your model

First time slicing I get error

### Add your .zip here ⬇️

[CE3E3V2_Space Heater v3 - Body 1 - 140mm - Solid Top.zip](https://github.com/Ultimaker/Cura/files/12648806/CE3E3V2_Space.Heater.v3.-.Body.1.-.140mm.-.Solid.Top.zip)

GregValiant, 2023-09-18:
Thanks for the report.
I was unable to open the file as a project in 5.4.0.  The problem appears to have been with the "generic_PLA_175" material.

I brought the model into 5.4.0 and with my Ender 3 Pro active and using my settings I didn't have any trouble slicing this.  The model doesn't appear to have any errors.

This might be a setting problem or it could be a bug.
There is a persistent bug in 5.4.0 that can keep models from slicing.  There are three things that individually or in combination can fool the bug.

- Rotate the model in Cura
- Move the model in Cura (not really an option with this model)
- Change the Initial Layer Height but a little bit (like from 0.20 to 0.23).

[image]

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic