Ultimaker/Cura #16761: Spurious "network enabled printers" entries in "settings > printer"

oliof, 2023-09-19:
### Cura Version

5.4.0

### Operating System

POP!_OS Linux 20.04

### Printer

RatRig V-Minion with Marlin

### Reproduction steps

click on Settings-Printers, see list of "network enabled" printers without names for the printers

### Actual results

[image]

At one time there were three of these, but clicking on them makes them go away

### Expected results

List should either not be populated, or be populated with actual names of the printers in question.

Please note that this printer has been running klipper before (Chi-Minion entry), and there are also RepRapFirmware printers configured (CR-Cast), while M-Minion runs Marlin. I have the RepRapFirmware and Octoprint Plugins installed. I did use the Klipper plugin in the past but uninstalled it because it interefered with the non-klipper printers (that will be another bug).

### Add your .zip and screenshots here ⬇️

[cura-networkedprinters-log.zip](https://github.com/Ultimaker/Cura/files/12662144/cura-networkedprinters-log.zip)

PatrickGlatz, 2023-11-28:
Any updates here? I do have the same issue and don't know how to delete them

Beep6581, 2023-12-09:
Same issue in Manjaro using Cura 5.6.0.

[image]

I see multiple entries in that list, and none of them have names.

Unexpected behavior follows when I click on an entry: it could be toggled on/off, or another entry could be toggled on/off but not the one I clicked, or nothing could happen.

I assume a new entry gets created based on the `/dev/ttyACMx` port.