Ultimaker/Cura #16766: Cura not start

Thibaut-tech, 2023-09-19:
### Cura Version

5.1.0

### Operating System

Windows 10

### Printer

Wanhao D12 230 Pro

### Reproduction steps

1. Start Cura

### Actual results

Nothing, cura just not start since 2 days

### Expected results

Cura start 

### Add your .zip and screenshots here ⬇️

[image]

kcoffey1959, 2023-09-23:
Mine started doing the exact same thing. Two days ago I had a lock up on my computer and upon restarting Cura would load no farther than this screen. I've tried repairing and reloading with no change to the situation. 
The only difference is I'm running cura 5.4

Thibaut-tech, 2023-09-24:
i think im gonna move to prusaslicer beacuse i have this issue too often with cura