Ultimaker/Cura #16768: "Build Plate Initial Temperature" only temperature setting that is not a float

oliof, 2023-09-19:
### Cura Version

5.4.0

### Operating System

POP!_OS Linux 20.04

### Printer

RatRig V-Minion with Marlin

### Reproduction steps

Open Settings, and tune print temperatures.

### Actual results

"Build Plate Temperature Initial Layer" shows as an int ("100") while all other temperatures show as floats ("100.0").

### Expected results

Temperatures shown should be consistent.

### Add your .zip and screenshots here ⬇️

[image]

5axes, 2023-09-20:
I not a Linux user, so I cannot help you on this point. But on Windows you can press the dot key and 0 to get the decimal value.
I hope this helps.
[image]

GregValiant, 2023-09-20:
Thanks for the report.
Since you did not include a project file I must surmise some things.

Within "fdmprinter.def.json" is:
"material_bed_temperature_layer_0":
                {
                    "label": "Build Plate Temperature Initial Layer",
                    "description": "The temperature used for the heated build plate at the first layer. If this is 0, the build plate is left unheated during the first layer.",
                    "unit": "\u00b0C",
                    **"type": "float",**
                    "resolve": "max(extruderValues('material_bed_temperature_layer_0'))",
                    **"default_value": 60,**

Printer Definition files and Material Files will also often send the temperature to the box as an integer as very few printers can discriminate to 1/10°.
If you follow @5axes suggestion (if it works on Linux) you can input up to three decimal places and it will display as such (Ex: 60.001).  If you require finer tuning than 1/1000 of a degree then you will need to manually change the numbers in the Gcode by using Search and Replace or some other post processor.
The current maximum resolution of the "Bed Temperature Initial Layer" setting box will show up in the gcode as:
M82
M140 S60.001
M109 S210

Using "Search and Replace" it would be possible to get a finer resolution of the temperature:
Search = M140 S60
Replace = M140 S60.00000000000000000000000000000000000000000000000000000001
Use Reg Exp: not checked

That would result in:
M82
M140 S60.00000000000000000000000000000000000000000000000000000001
M109 S210

I'm going to remove the bug label from this as it is "expected behavior" that the default number as called out in fdmprinter.def.json is an "int" although the box is designed to hold a float.  Other definition files and any custom material file may be sending the number over as an integer as well (when in fact it is indeed a float).  I addition, when I type "60" into the box it displays as "60.0" in which case I would say that @5axes workaround might be considered overkill (if it indeed works on Linux).
I hope you understand.

closed 2023-09-20 by GregValiant