Ultimaker/Cura #16769: when using support the first layer does not contain all of the structure needed

Johnep101, 2023-09-19:
### Cura Version

5.4.0

### Operating System

Mac  os monterey 12.6.9

### Printer

Ender 5 pro

### Reproduction steps

1) design in open scad exported .stl file
2) added support structures in cura

### Actual results

this is the first printed layer

[image]

this is the second printed layer, note the support layer being printed in thin air.

[image]

### Expected results

the first layer should contain the second layer support structure.

### Add your .zip and screenshots here ⬇️

project.3mf.zip
[top_wall.stl.zip](https://github.com/Ultimaker/Cura/files/12666511/top_wall.stl.zip)

GregValiant, 2023-09-20:
Thanks for the report.  We like to get a project file instead of the STL as a project file has your settings and printer in it.  Use the "File | Save Project" comand to create one.
I have to guess at your setup but this looks like a previously reported bug.  If the "Bed Adhesion" is set to "None" the Support Brim doesn't form and yes, the support is left hanging in the air on the second layer.
[image]

When the "Bed Adhesion" is set to anything other than "None" the support brim forms.  Here I've set it to Skirt with the Skirt Line Count set to "0".
[image]

This appears to be a duplicate of #14810 and a few others.  Let me know if the above fix works for you.
I'm going to go ahead and mark this as a duplicate and close it but if your problem is different it can be reopened.

GregValiant, 2023-09-20:
This should be fixed in the next release.  Good that you got it squared away.

closed 2023-09-20 by GregValiant