Ultimaker/Cura #16770: Allow Extra Skin Wall Count to be negative to remove Inner Walls

ZedOud, 2023-09-19:
### Is your feature request related to a problem?

I'd like to be able to eliminate extra walls only on the top surface so that the fill pattern reaches the Outer Wall.

### Describe the solution you'd like

Extra Skin Wall Counts already use the Inner Wall speed, despite using the Top Surface Skin Line Width for its width, as mentioned in my issue #16691

Thus, it follows that this setting should be able to affect the Inner Walls, at some level being considered an Inner Wall already.

### Describe alternatives you've considered

One could just print fewer Inner Walls and replace them with Extra Infill Wall Count.

### Affected users and/or printers

I think this will affect everyone as manipulating the Top Surface is one of the most common go-tos for quality improvement.

GregValiant, 2023-09-20:
Like this?  The model on the left has a mesh modifier configured with 0 Extra Skin walls and 1 regular wall on just the top layer.  The layers below have 3 walls like the model on the right.
[image]

ZedOud, 2023-09-21:
> Like this? The model on the left has a mesh modifier configured with 0 Extra Skin walls and 1 regular wall on just the top layer. The layers below have 3 walls like the model on the right.

Yes, exactly like that. Ideally this could be applied on the first layer as well, but the top skin layer would have the most effect.

5axes, 2023-09-21:
Similar to **Visible Top Skin Has Only One Contour**  : https://github.com/smartavionics/Cura/issues/183

GregValiant, 2023-09-21:
_Very_ similar.  If @smartavionics would include an "Extra Skin Wall Count" into the Top Skin Layer settings (maybe as "Visible top skin extra wall count" and "bottom skin extra wall count") it would be right-on the money.

5axes, 2023-09-21:
This is already the case with the "**Visible Top Skins Has Only One Wall**" and "**Bottom Skin has Only One Wall**" parameters. Although you can't define the number of contours (only one). That's the goal of this option.

[image]

GregValiant, 2023-09-21:
Ok I see now.  He combined the settings so that "Visible top skins have only one wall" is doing exactly what it says with no need to mention the "Extra skin walls" at all.  Nice.