Ultimaker/Cura #16771: Slicing gets stuck, CPU pegged at 100%; never completes

john94501, 2023-09-19:
### Cura Version

5.4.0

### Operating System

Mac OS Ventura 13.3.1

### Printer

Creality Ender 5

### Reproduction steps

1. Import STL file
2. Return settings to defaults
3. Click slice button

### Actual results

Slicing progresses initially and then freezes.

[image]

### Expected results

Was expecting slicing to complete the way it normally does :)

### Add your .zip and screenshots here ⬇️

[Slicer_Freezes.zip](https://github.com/Ultimaker/Cura/files/12667324/Slicer_Freezes.zip)

GregValiant, 2023-09-20:
Thanks for the report and the file.
Yes, this takes a while to slice in 5.4.0.  Maybe you just didn't wait long enough?

Slicing times for comparison:
- 00:02:30 in 5.4.0
- 00:00:15 in PrusaSlicer
- 00:00:03 in Cura 4.13.1

So it took 50X longer to slice in 5.4.0 than it does in 4.13.1.  This has been noted before in regards to 5.4.0.  Hopefully something will be done about this in the next release.  The Cura team will take a look.

john94501, 2023-09-20:
Hello Greg,

I waited longer than that, but perhaps my laptop is not as fast as yours.
It was pegged at 100% with fans running full speed for several minutes. In
the end, I played with the number of holes in the mesh until it sliced
faster (a few seconds again once I dropped it to 10 x 7 and removed the
side triangles & diamonds).

Thanks for the quick response,
John...

GregValiant, 2023-09-20:
When I stood it on one edge it sliced right away.  I do that with some models to see if slicing from a different angle makes any difference.  In this case it was a huge difference.

john94501, 2023-09-20:
Hello Greg,

I wondered about that but didn't try it because this mesh is actually part
of a larger model (I isolated it as the cause of the slow slicing by
breaking it into pieces) and printing the whole model on one edge might
have been a challenge. I had also considered whether I could make sure the
holes were perfect squares and then rotate by 45 degrees to align them on
the XY axes. The generator for that mesh doesn't have good controls for
ensuring they were perfect squares, so would probably have meant me
manually creating them in the model.  Interesting how the same model at a
different angle can make such a big difference though. Worth remembering if
I have future models that do similar things.

Rgds,
John...

GregValiant, 2023-09-20:
I don't know the inner workings of Cura, but the three main workarounds for the 5.4.0 slicing bug all have one thing in common, they change the angle that the slicer views the model.  So it appears that Cura's "view" of the model has some relevance to the bug.
That is a good model with no discernable errors and it should slice with no problem.

GregValiant, 2023-09-29:
@MariMakes and @fieldOfView this is the model with the high slicing time in 5.4.0.

GregValiant, 2023-09-30:
"Optimize Wall Printing Order" enabled 2:30, disabled 0:15.

closed 2023-11-16 by MariMakes