Ultimaker/Cura #16775: Slicing failed with tree supports enabled.

jallen128652, 2023-09-20:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Custom Tronxy X5s

### Name abnormal settings

Used default fine settings with generic ABS settings, enabled tree supports. 

### Describe model location

center x/y small model.

### Describe your model

articulated octopus file from thingiverse

### Add your .zip here ⬇️

Nothing special about this print, even printed the same file yesterday. The only change was using the default fine settings and tree supports. Normally I use custom settings.

GregValiant, 2023-09-20:
Thanks for the report but without a project file ("File | Save Project" and zipped) we have nothing to look at to figure out what might be going on.

jallen128652, 2023-09-20:
[CFFFP_Octopus_sup_v6.zip](https://github.com/Ultimaker/Cura/files/12674198/CFFFP_Octopus_sup_v6.zip)
Greg please see attached

GregValiant, 2023-09-20:
Thanks for the report.
I can't slice this with 5.4.0.  Prusaslicer was able to slice it.  As you can see here that slice is totally wrong.
[image]

This is in 4.13.1 and "appears" to be OK.  If it were a practical part I don't know if I'd trust it to print correctly.
[image]

I'll mark this as a duplicate slicing error as even after I repaired the defects in the model it still wouldn't slice in 5.4.0.

closed 2023-09-20 by GregValiant

jallen128652, 2023-09-20:
Greg it sliced fine and printed with my custom settings using cura 5.4 yesterday and again this morning. Just didn't work with default fine setting with tree supports enabled. 
I uploaded some photos here:
https://drive.google.com/drive/folders/1t88atR7ElNtMdJaqzVxeczrbT8GFQhm9
When I'm back at my desktop I'll zip you the project with my settings that does slice so you have a point of comparison if you want to pursue the issue further. 

Thanks

Jim