Ultimaker/Cura #16777: Selecting Build Plate Adhesion Extruder visibility does not add it to the print settings

GADify, 2023-09-20:
### Cura Version

5.4.0

### Operating System

MacOS Monteret 12.6.1 AMD Radeon Pro W5700X 16GB

### Printer

Ultimaker S5 Pro

### Reproduction steps

In Print Settings, under Build Plate Adhesion, click the three lines to pull up the preferences visibility window. 
Select Build Plate Adhesion Extruder

### Actual results

Nothing happens

Note that if I deselect Build Plate Adhesion Type then that is immediately removed from Print Settings and I can add it back, but Build Plate Adhesion Extruder will not show. 

### Expected results

Build Plate Adhesion Extruder should show up in Print Settings

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2023-09-20:
Thanks for the report.
Extruder#2 looks to be grayed out as if it was disabled.  That would cause Cura to consider your S5 as a single extruder printer and there won't be options for a second extruder under any of the categories.
With both extruders enabled I can't duplicate this problem.

GADify, 2023-09-20:
Sorry - totally my fault. Thanks for the quick reply. Spelling errors courtesy of iPhone

GregValiant, 2023-09-20:
Not a problem.
I'll go ahead and close this then.

closed 2023-09-20 by GregValiant