Ultimaker/Cura #16778: Auto-tower file name too long for some printers

njefferson, 2023-09-20:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Flashforge Dreamer NX

### Reproduction steps

Extensions-->Auto Tower-->Custom Retract Tower

### Actual results

This results in a filename like this:  "Custom Retraction Tower - Distance 1.0-6.0x1.0"
This is too long for my printer to read.

### Expected results

My Anycubic Kobra Neo has no problem with it, but my older Flashforge Dreamer NX cannot open the file until I rename it to something shorter.  I do not know the maximum character limit, but it works when I make it much shorter.  Special characters do not seem to matter.

### Add your .zip and screenshots here ⬇️

n/a - the file will not load on the printer

GregValiant, 2023-09-21:
Thanks for the report.
There are two Calibration plugins available in the MarketPlace.  One is "Calibration Shapes" by @5axes and the other is "Auto Tower Generator" by [Brad Kartchner](https://github.com/kartchnb).  You may want to post this bug report on the relevant Github page as they are the maintainers of their plugins.

njefferson, 2023-09-21:
Oh, thank you very much.  I will do that.

GregValiant, 2023-09-21:
I'll go ahead and close this.  Hopefully he'll get it fixed but a downside to fixing this is that it will likely affect ALL printers.  That may be something that the author is not willing to do.

You can change the name of a project down in the left lower corner of the Cura workspace.  The bottom line of the "Object List" is editable so you could change it there before you save the file.  Because there is a method in place to shorten the name in Cura I'll remove the bug label from this.
Another workaround would be to save the gcode to the hard drive, shorten the name, and then copy the file to the SD card.  That's a few more steps but it would work.  That is in fact how I always do it so I can keep the gcode files in a proper folder.

closed 2023-09-21 by GregValiant