Ultimaker/Cura #16779: Unexpected Dimention Change

AbsolutAI, 2023-09-21:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Creality Ender 3 V2 Neo - Professional Firmware BLTUBL MPC 20230904 for 4.2.2 stock board

### Reproduction steps

Steps to replicate:
 1. Imported STL file.
 2. Verified dimensions via Scale ("S") were 100% and took note of the reported X, Y and Z dimensions in mm.

[image]

 3. Added support blocker ("E"). Verified that the Scale tool still displayed the same measurements as before. 

[image]

 4. Used Per Model Settings on the added support blocker.
 5. Selected "Normal Mode" as mesh type.
 6. Moved the support blocker from where it landed when imported to a random location, using the Move ("T") tool. For demonstration purposes, the support blocker landed on X(-27.2477), Y(-11.7217), Z(11.68).
 

### Actual results

7. After displacing the imported support blocker, dimensions of the imported STL file were different when seen in the Scale ("S") panel, despite the model not changing its actual dimensions on the Cura environment.

[image]

Culprit confirmation:
 8. Repositioned the support blocker aproximately where it originally landed, by displacing it with the mouse.
 9. Reported dimensions for the STL in the Scale ("S") panel were back to their original value from when the STL just got imported.

[image]

 10. Manually resetting the dimensions of the STL to 100% now changes the actual dimensions of the model, even when stating it is back to its original X, Y, Z dimensions; hence, it prints innacuratelly and deformed since the slicer view shows the deformed part sliced as is.

[image]

 11. Right-clicking the STL and selecting "Reset all model transformations", as it's named, resets the STL transformations along with the support blocker's, reverting all issues.

Only when the blocker leaves the volume of the model, does it start to alter its "dimensions" in the Scale panel, in proportion to how much the blocker pertrudes or is far away from the model.

For X
[image]
_Support blocker pertruding from the model in the X direction_.

For Y
[image]
_Support blocker pertruding from the model in the Y direction_.

For Z
[image]
_Support blocker pertruding from the model in the Z direction_.

### Expected results

Expected results should be those seen with other mesh types for the support blocker cube, that is, no change in the model's dimensions whatsoever, be it in the Scale panel or the actual dimensions/geometry.  Moreover, I'd expect the same behaviour as if I was importing any other aditional model into the virtual print bed. Hence, I think it's a bug that needs some Raid and squashing.

[image]
_Support blocker outside model's volume. No change seen in X, Y, Z units. Mesh type: Modify settings for overlaps_

### Add your .zip and screenshots here ⬇️

Normal Mode and Print as Support mesh types show the same behavior. Reverting the support blocker to the default "Don't support overlaps" and reloading the model via "Reload all models" resets the dimensions shown in the Scale panel too.

[image]
_Mesh type: Modify settings for overlaps. No change in dimensions._

[image]
_Mesh type: Don't support overlaps. No change in dimensions._

[image]
_Mesh type: Print as support. Same issue presented._

GregValiant, 2023-09-21:
Thanks for the report.
When you add a support mesh to a model it becomes part of the model.  Whether you have it configured as normal model, or overlap modifier.  As a Support Mesh or Support Blocker it's treated differently.
The model "size" dimensions consider the size of the "bounding box" of the model.  The bounding box does not rotate with the model but rather is always square to the world.  Therein lies the effect you are seeing because when you move the support mesh outside the main model you are making the "bounding box" bigger, not the model itself.
Your model never changes size in Cura but the bounding box does.
You can see the same effect without the blocker in the view.  Simply rotate your model.  If you happen to have a 25mm cube on the build plate and you rotate the model 45° you will see the "Scale" boxes change and reflect a size of 25/SIN(45) or 35.35.  That is the X and Y dimensions of the bounding box which is the X and Y edges of the space that the model takes up on the build plate.

So this is expected behavior.  Your model is fine and it's size is not changed.  The footprint on the build plate changes depending on where you move that block to because Cura considers it part of the model that was selected when the block was brought in.  When you move the block you will also see that the "part origin" moves to remain at what is the Center of Geometry of the two pieces.

This image is of two cubes and I've "Grouped" them and rotated the group in order to view the bounding box.  The gray box around the models is the bounding box and you can see that it doesn't describe the size of the models (25 x 50), but rather the area they take up on the bed.
[image]

I'll remove the bug label from this but I'll leave it open for a bit so you can respond.  I hope you understand.

AbsolutAI, 2023-09-22:
Oooooh. Yes, I did see the same behaviour when rotating objects and had some suspicions about how Cura managed the scale measurements, but never thought it was actually the "volume" that encloses the model, or as you name it, the size of the bounding box. With that into consideration it all makes perfect sense now. In any case, it is even more evident to me now that a certain workflow must take place so as to avoid the mistakes I made regarding the dimensions of my model.
Thanks for the information and quick reply.

GregValiant, 2023-09-22:
I'll go ahead and close this then.

If you want to see some strange behavior, rotate something to 45° and then turn off "Uniform Scaling" and change the model size by pulling on the X or Y grip.  You can see that the bounding box changes size, but the model inside gets pulled out of shape.
A rotation must always be the last transformation you make to a model.

closed 2023-09-22 by GregValiant

AbsolutAI, 2023-09-23:
I did notice that at some point some weeks ago (since I'm pretty new to 3D printing anyways). I got really bamboozled when I tried to resize a cube after rotating it. For better or worse, I did find out a rotation must be the last step always xD