Ultimaker/Cura #16789: [5.4.0] Poor / inconsistent slicing of narrow ridge

rcls, 2023-09-21:
### Cura Version

5.4.0

### Operating System

Linux (Fedora 38)

### Printer

Creality Ender-3 S1 Pro

### Reproduction steps

The attached model has four identical narrow sloping ridges. 3 slice sensibly, but one (+x+y) has odd artifacts. Different layers get slightly different approximations to the ridge, resulting in overhanging segments, and at the top, a completely detached floating piece.

Rotating the model by 180° (or mirroring it), it is always the edge at positive x and y print bed position that has the artifact, so it seems clear that this is a slicing issue and not a defect in the model.

Rotating the model by 90° the slicing is much better.

The attached image is looking at the poorly sliced ridge from the side, the artifacts are obvious. Cura project .3mf is also attached.

[image]
[ti2.3mf.zip](https://github.com/Ultimaker/Cura/files/12692173/ti2.3mf.zip)

### Actual results

Slicing the model gets artifacts on one of four identical edges.

### Expected results

It would be much better if all four edges had similar quality to the good three.

### Add your .zip and screenshots here ⬇️

[image]
[ti2.3mf.zip](https://github.com/Ultimaker/Cura/files/12692173/ti2.3mf.zip)

GregValiant, 2023-09-21:
There is something wrong with the zip file you posted.
Just to make sure, load the model into Cura, set Cura up to slice, and use the "File | Save Project" command.  Go to the folder where the 3mf was saved and zip the file.

That sure looks like a feather edge that is simply too narrow to slice with your nozzle and settings.  Try dropping your line width  and minimum line width to 0.20 and slice it just to see if more of the edge gets filled in.

rcls, 2023-09-21:
Attempting to upload the project again. .zip file containing a .3mf file.

[ti2.3mf.zip](https://github.com/Ultimaker/Cura/files/12693514/ti2.3mf.zip)

rcls, 2023-09-21:
> That sure looks like a feather edge that is simply too narrow to slice with your nozzle and settings. Try dropping your line width and minimum line width to 0.20 and slice it just to see if more of the edge gets filled in.

It sure is pretty narrow. However, that doesn't explain why three other identical edges get printed much better.

Dropping the line width and minimum line with to 0.20, I get similar artifacts, just smaller:
[image]

GregValiant, 2023-09-21:
This is with a 0.10 line width
[image]

Those edges become "infinitely thin" and simply aren't suitable for FDM printing.  They need a flat area or they aren't going to slice correctly.
Here it is in PrusaSlicer.  It shows the same inability to slice those razor edges.  This print would fail.
[image]

I'll leave the bug label on this but it looks to me like the model needs an adjustment.

rcls, 2023-09-21:
@GregValiant you are missing something. There are four identical edges, and one is much worse than the others (the other 3 print acceptably).