Ultimaker/Cura #16792: Need dark mode to easy eye strain.

Chuckling, 2023-09-22:
### Is your feature request related to a problem?

Related to user discomfort.

### Describe the solution you'd like

N/A

### Describe alternatives you've considered

N/A

### Affected users and/or printers

Large bright white screen are had for user to sit in front  of for any period of time.

### Additional information & file uploads

N/A

5axes, 2023-09-22:
We already have a dark mode in Cura. **Preferences -> Configure Cura -> Theme** :  
## Ultimaker Dark

[image]

## Ultimaker Darkish 

[image]

##or 
[image]

GregValiant, 2023-09-22:
@Chuckling  - Did you find the setting?

closed 2023-10-07 by github-actions[bot]