Ultimaker/Cura #16794: Slicing failed with a regular stl from Printables : on Win 11 desktop PC

lwiart, 2023-09-23:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Bambulab X1C

### Describe your model

Model from printables : https://www.printables.com/model/563912-safe-from-the-rain

### Add your .zip here ⬇️

Just dropped the model on the plate and pressed "Slice" -> got message popup "Slicing failed"
[CFFFP_Safe from the Rain.zip](https://github.com/Ultimaker/Cura/files/12706143/CFFFP_Safe.from.the.Rain.zip)

GregValiant, 2023-09-23:
Thanks for the report.
That is a truly interesting model.  I didn't have any trouble slicing it so you may have run into the 5.4.0 slicing bug.  There are three workarounds that can fool the bug.

- Rotate the model in Cura
- Move the model away from the Cura build plate midpoint
- Change the Initial Layer Height by a little bit.

I'll mark this as a duplicate and @MariMakes of the Cura team will take a look.  Let us know if one of those works for you.

lwiart, 2023-09-23:
Hi Greg,

Thanks for the workaround proposals : rotated model, moved model on build plate, tried several layer heights and initial layer heights (0.12, 0.16, 0.2), all result in the "Slicing failed" popup.

I'm wondering if this is because of my PC hardware, because I get the same issue with Bambu Studio (it crashes during slicing)...
I've : 
- Win 11 10.0.22621
- Asus ROG Maximus Z790 Hero
- CPU i9-13900K
- Corsair Vengeance DDR5 6600 MHz
- Nothing overclocked though

lwiart, 2023-09-23:
Extra remark (very interesting IMHO) : 
- I've a 0.2 nozzle installed on my printer
- So I slice at 0.2mm layer height (draft) -> "Slicing failed"
- I change to 0.4mm layer height (coarse) -> slicing goes through, no "slicing failed"

Hope this helps

GregValiant, 2023-09-23:
The model has some problems.  I didn't look closely at it because it sliced for me with no problem.
This is from formware.co/OnLineSTLRepair.  They have a file size limit but I like there report:

-> Reading file and indexing vertices
-> Analysed your file:
--> 0 Naked edges (?)
--> 0 Planar holes (?)
--> 0 Non-planar holes (?)
**--> 72 Non-manifold edges (?)
--> 112 Inverted faces (?)**
--> 0 Degenerate faces (?)
**--> 6 Duplicate faces (?)**
--> 0 Disjoint shells (?)

That's not a bad report but there is never any way to know if a model error has caused Cura to hang, or if it's a bug.
The problem seems to be more prevalent in Win11 computers but that's an un-scientific guess on my part (but I do see al the bug reports).
The normal workarounds are to Rotate and/or move the model in Cura and/or change the "Initial Layer Height" by just a bit.  You may find that going from a layer height of .20 to .21 can fool the bug as well.

lwiart, 2023-09-23:
Right, It crashes a lot on Win 11.
Interestingly enough, I installed a WIndows 10 VM on my Win 11
(VirtualBox), and there is no crash whatsoever when slicing with the same
params I try to slice with on my Win 11 install of Cura...

GregValiant, 2023-09-23:
I may print this.  It isn't an easy one (and it's not a Mandolorian helmet).
It is certainly an excellent test for the Bridge settings.

lwiart, 2023-09-28:
Hi GregValiant,

I got to the bottom of the problem.
It wasn't related to Cura but to a problem with my ASUS motherboard (actually many recent ASUS motherboard models) and my i9-13900K.

A parameter in the Bios has to be changed : SVID Behavior (change from ""auto" to "Intel's Fail Safe").

Now Cura doesn't crash when slicing.

GregValiant, 2023-09-29:
@rburema ,
The OP's fix sounds interesting.

lwiart, 2023-09-29:
Quick note : seems that there are issues with many Intel core i9-13900 (versions K/etc.) that are kind of unreliable in multi-thread intensive tasks.
So what I really meant is that this "unreliability" of the cpu can be mitigated with some Bios parameters.

Interestingly enough in my case : Cura, PrusaSlicer, Bambu Studio would all systematically crash during slicing of some specific models at low layer height (like <0.30/0.35) when this SVID Behavior param in BIOS was set to "auto" and not "Intel's Fail Safe".

I'm not an overclocker, nor into optimizing CPU/motherboard parameters (there are 10s of parameters related to frequency, voltage, etc. and I've no clue what they are meant for...), but in my case, it was just a parameter set by default in the Asus motherboard BIOS that would reveal this potential "unreliability" of the i9-13900K processor.

closed 2023-09-29 by lwiart

rburema, 2023-10-04:
@GregValiant -- That _does_ sound like an interesting fix, but unfortunately not something we can (or should) be able to set from Cura. ( 'User level-program changes users' BIOS-setup' is not a headline I'd want to ever read ;-) )

We've been having similar problems in the past, but I'm in a way 'glad' that the others' are also struggling with this problem, as it shows that it's very likely an Intel and/or ASUS problem, rather than on our side. -- Not that his helps the user unfortunately :-/

lwiart, 2023-10-04:
Thanks for weighting in rburema. Definitely not a Cura problem :) 

My take away is that Cura and the other slicers I've tried (PrusaSlicer, Bambu Studio) are amazing multi-thread "stress tests" for CPUs : when there is a problem with multi-core CPUs, the slicers crash in less than 10 secs.

So you might see similar posts in the future unfortunately, because as far as I have seen out there (posts on forums), many people have problems with the "high end" Intel CPUs of the 13th generation, and the slicers are amazing at spotting this.

GregValiant, 2023-10-04:
I came across another article (or post) that described problems with "over-clocked" CPU's.  I'm not savvy enough to know if that is an indicator to what might be going on but I thought it was intriguing.

lwiart, 2023-10-04:
If you have the pointer, I'm interested ;)

My (limited) current undestanding, since I'm not into OC and just want a powerful and stable PC is:
- on MB side, there are "native" overclockings that are turned on by default  by some MB vendors (but hard to know and find without a fair amount of digging). Example, in my case with an Asus Z790 board : **BIOS MultiCore Enhancement** enabled -> PC unstable (CPU).

- on CPU side (and I'm talking for my specific case with i9-13900K) : there is a difference between the "marketing promise" that says that K versions of CPU are overclockable and the ground truth where it's more of a "silicon lottery" and you don't know what you get. Example, in my case -> probably not a great CPU, so I have to set **SVID Behavior** to "Intel's fail safe" because my CPU needs to be overvolted to be stable with multi-thread hungry tasks...
 
That being said, I'm not an overclocker and just kept all BIOS params default from the beginning --> PC unstable with slicers.
Until I dug a little more...

GregValiant, 2023-10-04:
I don't keep good track of where I've been when I have to dig into these bug reports.  I'll go through my history, but this isn't my only PC so it will likely be fruitless.  I will do it though.

rburema, 2023-10-05:
@GregValiant The short version is; Motherboard vendors are overclocking the CPU's they put in, when they shouldn't, and there's little to nothing Cura (or other slicers) can do about it. Well, unless we want to slice slower -- there is (I _think_) still a way to get the engine to slice with only 1 thread, but it requires jumping through a few command-line hoops (starting the engine manually, first signal Cura that you want to with `--external-backend`, then use the engine from the command-line every time you want to slice, and add `-m1` to the usual command line parameters to start with only 1 thread.). And it'll be very slow.

lwiart, 2023-10-05:
On Windows, simple solution : launch Cura, open Task Manager, "Details" tab and find the cura process, right click, "set affinity" option, then chose which cores to use (can be 1 or several)

GregValiant, 2023-10-05:
"Set Affinity" eh?

@lwiart do you happen to know if the default is always "All Processors"?
We don't want @rburema on the front page in bold type but if this is something that could vary by system, could it be picked out and added to the log?
I don't see any operating system information in the log.  Is it possible it could have some value down the road if it were added?

lwiart, 2023-10-05:
Yes, default is always all the CPUs when an app/process is launched on Windows.
In fact, this is rather "threads" than CPUs, to be correct.

For example : 
- On my i9-13900K, there are 24 cores (8 P-Cores 2 threads, 16 E-Cores 1 thread)
- When I "set affinity" on Windows for a process, by default, **UC0 to UC31** are all selected.  I can then unselect checkboxes (1 per UC)
- I guess (but haven't verified) that UC0 to UC15 are for the P-Cores, and UC16 to UC31 are for E-Cores. But I can't be 100% sure, I didn't verify that my assertion is true.

rburema, 2023-10-05:
@GregValiant 

We could add something to the logs perhaps, yes, if we can find a way to get that information from inside Cura -- I'll add it to the board.

As for the other questions:

If the default wasn't 'all processors' (or similar), then Cura would limited in the ability to multi-thread (so, divide tasks over different cores -- since modern PC's have more than 1 CPU basically). Cura (mostly the engine) make extensive use of multi-threading, since you can do a lot of stuff per layer, independent of the other layers -- this speeds the process up considerably. (Also why slicing on Mac was very slow a few years back, since we only had one 'thread' in the engine then.)

We have no way of detecting whether a slice-error happens because of a fault by overclocking, since when something goes wrong with that, almost anything can happen. On the other hand, this may be a quite common tactic by vendors, so that warning people beforehand would tag so many people that the warning would become useless (if we could even find a way to detect those settings from inside of Cura). -- But again, like you said, perhaps adding this to the logs would help us.

GregValiant, 2023-10-05:
It was just a thought.
The percentage of "won't slice" reports has become heavily weighted towards Win11 computers.  That may well be because there are more of them now (and they are newer machines).

If it was possible to retrieve the info from Windows it might be helpful down the road.  Constantly tweaking the Engine to solve what may turn out to be a hardware problem may be an exercise in futility.  I have this feeling that it could be helpful to be able to pull some:  Operating System, Motherboard, CPU, and GPU data and stick it into the log.

The "won't slice" problem is why I opted for a Win10Pro laptop rather than a Win11 machine.  There have been a lot of reports where I have to respond with "Well it sliced fine for me on Win10".  I've even been accused of using a "special" version of Cura that works better.

lwiart, 2023-10-05:
Lol :)

Nope, it seems to be more related to recent MBs and CPUs. Even more so when
some MB vendors overclock the CPU by default.
That, with CPUs vendors selling CPUs as overclockable whereas it's more of
a silicon lottery (like in my case, where I have to slightly over volt the
CPU for it to be stable in multi-thread even though it's not
overclocked...), and you get the perfect recipe for tons of "bug reports"
since slicers exploit multi-threading....

Ergonomicmike, 2023-12-05:
FWIW, Safe-from-the-rain did not slice correctly for me either, either in 5.5 or 5.6.

In the screen shot from 5.6, Standard Quality, you can see 1) the bridges don't print. (Cura shows black lines.) 2) the "Cura bug" (like my Taco bell bug) where there's an extraneous line.

[image]

GregValiant, 2023-12-05:
I can get it to slice with the Minimum Line Widths dropped to 0.28.  I tried printing this as I had some leftover glo filament that I thought would look pretty cool.  It was a bit damp. so the print didn't go so well.
One problem was that the ends of the "rain" aren't connected to the walls.  The extrusions don't want to start.  Some of them completed but others just fell down.  Here you can see that the ends appear to be just sitting in pockets rather than part of the model.
[image]

Ergonomicmike, 2023-12-05:
Thanks for the reply. I'm sure that you're correct about the bridge anchors. But, FWIW, printed fine using Prusa. (And I greatly dislike the unintuitive UI in Prusa.)

A subtle problem with the model is that it doesn't do well printing with a heated bed/plate. I printed in translucent PTEG. It printed fine. But the bridges near the plate sagged during printing from the heat of the bed. I don't see  a way around that if one needs a heated bed for adhesion.

GregValiant, 2023-12-06:
Even printing with the bed at 85, I get warpage from PETG (and I print mostly PETG).  I design elephant ears into most of my stuff but as the print gets tall it wants to pull the corners up.  Anything down low is going to be affected by the bed heat and I can see those "rain" features sagging.   That's a tough model.

You can see the dimples in the elephant ear below.  It was 2mm thick.  That area was on the bed and should be flat. 
[image]

Ergonomicmike, 2023-12-06:
Would Core XY mitigate the pulling of a model off the bed? (Am thinking no movement = less impetus to fail/less air movement to cause warpage.)

GregValiant, 2023-12-06:
An enclosure would probably take care of it.  That's about it though.  As the print gets taller the shrinkage in the XY just starts pulling the corners off the bed.  PETG is much like ABS in that regard.  Keeping the upper portions closer to the glass temperature would be the best.  When the print is removed from the plate everything could shrink at the same rate.  
The bonding to the bed is necessary but it's also part of the problem since at the bed the print can't shrink in the XY.  I've gotten pretty good at recognizing the shapes that are prone to warping (long narrow pieces are bad) so I make adjustments to the model in an effort to hold the edges down long enough for the upper portions to complete.  There is a plugin in the MarketPlace for Anti-Warp tabs.  Trimming off whatever shape is used for elephant ears doesn't take but a minute.