Ultimaker/Cura #16795: Ender-3 v3 SE Profile missing

ProTechEx, 2023-09-23:
### Is your feature request related to a problem?

Unfortunately cannot work with Cura because Ender-3 v3 SE Profile is missing

### Describe the solution you'd like

Adding Ender-3 v3 SE Profile

### Describe alternatives you've considered

I've try to a add Ender-3 v2 Profile, but I notice that the Printer Setting are different in comparison.

### Affected users and/or printers

Creality Ender-3 v3 SE

andytimofeev, 2023-09-28:
Thank you for adding this feature request - also looking how to solve the problem with printing settings for Ender 3 V3 SE

eborchardt, 2023-10-02:
I'd also like to see a profile for the Ender 3 v3 SE.

evanlloydjohns, 2023-10-06:
I also hoping to see an Ender 3 V3 SE profile in Cura

marcvandecoevering, 2023-10-07:
yes, please!! :)

Murat0773, 2023-10-08:
please..

saumyaj3, 2023-10-11:
Hi! These changes will be available in the CURA version after 5.5. 
If you still wish to use the machine please follow the comment: https://github.com/Ultimaker/Cura/pull/16796#issuecomment-1748634700

closed 2023-10-11 by saumyaj3

WilliamSC65, 2023-10-26:
> Hi! These changes will be available in the CURA version after 5.5. If you still wish to use the machine please follow the comment: [#16796 (comment)](https://github.com/Ultimaker/Cura/pull/16796#issuecomment-1748634700)

Vraiment dommage que la dernière version (5.5) de Cura n'ai pas intégré le profil Ender 3 V3 SE 👎

joostvhts, 2023-10-26:
> Vraiment dommage que la dernière version (5.5) de Cura n'ai pas intégré le profil Ender 3 V3 SE 👎

Zoals je hier kunt lezen is er een manier om de profielen alsnog nu al te gebruiken 🙃

WilliamSC65, 2023-10-26:
> Zoals je hier kunt lezen is er een manier om de profielen alsnog nu al te gebruiken 🙃

Oui je l'ai installé dans Cura 5.4.0 et cela fonctionne parfaitement 👍

joostvhts, 2023-10-26:
> Oui je l'ai installé dans Cura 5.4.0 et cela fonctionne parfaitement 👍

Lekker gewerkt pikkestein en wat fijn dat we elkaars taal spreken dan hebben we die ene niet nodig die iederéén kan 🙃

jsleroy, 2023-10-28:
Had a try at the profile for my V3 SE.
This works well but there's stringing I can't get rid off.
I'm using a V2 Neo with no issues at all.
So probably this profile needs some refinement (working on it but no success yet).
In case, find attached the profile extracted from Cura slicer.

[v3se.cxprinter.txt](https://github.com/Ultimaker/Cura/files/13195501/v3se.cxprinter.txt)