Ultimaker/Cura #16798: Nozzle 1.2mm support please!

kris0725pl, 2023-09-23:
### Is your feature request related to a problem?

i purchase nozzle 1.2mm and cura max is 1.0 mm

Please add more nozzle support. 

### Describe the solution you'd like

add more support nozzle

### Describe alternatives you've considered

add more support nozzle

### Affected users and/or printers

none

Asterchades, 2023-09-23:
Cura itself has no maximum nozzle size. It will warn you at 10mm, but other than that you can give it whatever non-zero value you want, down to three decimal places.

If the drop-down list on your specific printer doesn't include it, you need to manually enter the details in the Machine Settings dialogue (Preferences -> Configure Cura -> Printers -> Machine Settings, in the Extruder tab). If you want it to be in the drop-down you'll need to create a new variant for your printer definition which sets the details for you, which you could then submit as a pull request for other users.

GregValiant, 2023-09-23:
It has always looked to me like Cura just uses nozzle sizes to hold some defaults (line width, layer height, etc.).

This is sliced with a 0.2 nozzle at 5.0mm line width and 2.0 layer height.  If there was actually a 5mm nozzle in the printer then this would work.
[image] 

The nozzle sizes are a function of the printer definition file and associated sub files.  Those files come from either a member of the community or the printer manufacturer.  It isn't an UltiMaker or a Cura thing.
So as @Asterchades says, there really isn't a limit on the nozzle size for any printer, just a warning at 10mm.  You just need to input the number in the correct place.

If it makes you feel better you can open your printer definition file in a text editor and see if it has variants.  If it does, then you can go to the "variants" folder in the Cura installation folder and open the nozzle file for your printer, change a couple of numbers, do a save-as, and there ya' go.
This is the location of the "variant" folder in my Windows installation:
"C:\Program Files\UltiMaker Cura 5.4.0\share\cura\resources\variants"

This is the total contents of the 0.4 nozzle for my Ender.  I just need to change two numbers and save the file under a different name using the same file naming convention.
[general]
definition = creality_ender3pro
name = **0.4mm Nozzle**
version = 4

[metadata]
hardware_type = nozzle
setting_version = 22
type = variant

[values]
machine_nozzle_size = **0.4**

I'm inclined to close this as it really isn't a thing Cura can even do.  It already supports 1.2mm nozzles.  Let us know what you think.

closed 2023-10-08 by github-actions[bot]