Ultimaker/Cura #16799: Failed Bracelet

Engrmpw1, 2023-09-23:
### Cura Version

5.4

### Operating System

Windows 10

### Printer

Ender 5 S1

### Describe your model

Bracelet

### Add your .zip here ⬇️

[CFFFP_Celtic_Knot_Bracelet_Medium.zip](https://github.com/Ultimaker/Cura/files/12707144/CFFFP_Celtic_Knot_Bracelet_Medium.zip)

Engrmpw1, 2023-09-23:
[image]

GregValiant, 2023-09-23:
Thanks for the report and the file.
I moved and rotated the model and it sliced.  Sometimes changing the Initial Layer Height can fool this bug as well.
[image]

I'll mark this as a duplicate and go ahead and close this as it is an often reported bug and the Cura team has lots of examples to work with.  Hopefully it will be fixed in the next release.  I hope you understand.

closed 2023-09-23 by GregValiant