Ultimaker/Cura #16800: Problems with saving part dimensions in a file when there is support blockers

SurfBlaster, 2023-09-23:
### Cura Version

5.4.0

### Operating System

Windows-10 x64

### Printer

Custom, like Prusa i3

### Reproduction steps

Firstly, this problem does not always occur, and I do not quite understand what it depends on.

When I resize the part a little (to compensate for shrinkage) and then add 'support blockers' to it, the dimensions on the screen are fine. But now it’s enough to save the file and open it again as a project, and strange %% sizes will appear that do not correspond to what I entered a couple of minutes ago.

### Actual results

[image]

Here's an example. The size in the top field is 100.7%, but I didn’t write anything like that there, it should be 100.5%.

### Expected results

I would like the files, after recording, closing, opening, to look exactly as if the session of working on the model had not been interrupted, without surprises.

### Add your .zip and screenshots here ⬇️

... I would like to add that it seems to me that the problem most often occurs when “support blocker” sticks out beyond the parallelepiped boundaries of the model.

[image]

Here the % fields should contain the values 100.5, 100.5, 100. Pay attention to the “z” axis, and to the high support blocker on the model.

SurfBlaster, 2023-09-23:
[image]

Sorry, wrong screenshot #2 !

GregValiant, 2023-09-23:
Thanks for the report.
The numbers shown in the "Scale" tool text boxes are dependent on the size of the "Bounding Box" that contains the model.  Since support blockers become part of the model (according to Cura) then they are part of the Group and so they count in the overall size.  If they protrude past the model itself, they look like they are making the model bigger.

After opening a project and seeing what is happening, try selecting the model, right-clicking and "Select All Models" and then right click and see if "Ungroup Models" is available (it might be grayed out and if so then hold "shift" and select each block and the model).
When models are Grouped the size of the group is noted in the boxes, not just the model itself.  This is fairly confusing, but it's the way it currently works.  If you rotate the model you will also see the sizes change because Cura isn't reporting on the model, it's reporting on the bounding box.
Here is a model and a support blocker.  The blocker is directly over the model but at around 140 in the Z.  The model is obviously not square and it is actually 24mm tall and not 150.00.
[image]
Here I've selected all models and grouped them.  Now the bounding box is visible and it indeed square and tall.  Notice that the Z is 100%
[image]
Here I've ungrouped the model.  Check out the model size.  Z is now 625% because that block way up high is making the bounding box taller than the model is.
[image]

In the Material Settings are three settings for shrinkage adjustments.  You might be better off using those setting rather than scaling.
I'm going to leave this open and the Cura team will take a look.  There may indeed be a bug here but rather than size being the issue (which has not been reported by anyone else) Cura may be wrongly "Grouping" all the meshes on the build plate when you save the project file and so things seem different when you open that saved project file.
At any rate the size issue appears to be "expected behavior" and yes, it is certainly confusing and especially so for people used to CAD programs.