Ultimaker/Cura #16802: Provide option to move, scale, mirror relative to selected model

rajinder-yadav, 2023-09-24:
### Is your feature request related to a problem?

Currently when using the move and scale operation, the X-Y-Z axis are relative to the build plate. The problem I am encounting is after I have rotated the model along with it's "support blocker" if I want to scale and move it, it is impossoble to make correction as the vector are to the build place and not  the model I've applided it to.

### Describe the solution you'd like

Would be nice to be able to have a check mark option to set axis relative to the last selected model.

### Describe alternatives you've considered

none

### Affected users and/or printers

general problem with slicer

rajinder-yadav, 2023-09-24:
[image]

rajinder-yadav, 2023-09-24:
Would like to see something like this. The same could be applied to mirroring as well as scaling and moving.

[image]

GregValiant, 2023-09-24:
Cura's UCS always stays square to the build plate and never rotates.  Right now it isn't possible to orient it in the manner you show in the screenshot.
I think one of the reasons for that is the UCS is always aligned to the bounding box of the model and not the model itself.  You can see that in the "scale" numbers when you rotate your part.  This box isn't square, but the numbers are reflecting the size of the bounding box and subsequently the area that the model takes up on the build plate.
[image]