Ultimaker/Cura #16815: [5.4.0] segfault on Open File(s) menu

oyvindaabling, 2023-09-25:
### Cura Version

5.4.0

### Operating System

Linux Debian 12.1

### Printer

Artillery Sidewinder X1 v4

### Reproduction steps

Start UltiMaker-Cura-5.4.0-linux-modern.AppImage, select File -> Open File(s) ..., or press Ctrl-O.

The legacy Linux version (UltiMaker-Cura-5.4.0-linux.AppImage) on a Debian 11.6 shows the same behaviour.

Cura complains in the logfile (attached) that:
WARNING - [MainThread] UM.Qt.QtApplication.__onQmlWarning [444]: file:///tmp/.mount_UltiMab2t2DI/share/cura/resources/qml/Cura.qml:624:5: QML FileDialog: Failed to load non-native FileDialog implementation:
qrc:/qt-project.org/imports/QtQuick/Dialogs/quickimpl/qml/FileDialog.qml:4 module "Qt.labs.folderlistmodel" is not installed

### Actual results

Cura segfaults with these messages:

QFSFileEngine::open: No file name specified
/tmp/.mount_UltiMadZb8FI/AppRun: line 25: 17528 Segmentation fault      $scriptdir/UltiMaker-Cura "$@"

If I instead choose File -> Open Recent -> xyz.stl, it works fine - the model loads, and I can slice, preview, save, etc.

### Expected results

The "Open File(s)" file selection popup window should appear.

### Add your .zip and screenshots here ⬇️

[cura.log](https://github.com/Ultimaker/Cura/files/12717801/cura.log)
[stderr.log](https://github.com/Ultimaker/Cura/files/12717805/stderr.log)

phtalo, 2023-10-08:
I want to report having the same bug.

Cura version: 5.4.0-1

Printer: Artillery Sidewinder X1

OS: Arch Linux 6.5.5arch1-1

Thank you for your time.

niknah, 2023-10-16:
The way around it is to open it up from the command line or file browser...  https://aur.archlinux.org/packages/cura-bin

Rushmore75, 2023-12-01:
Same Issue, slightly different error codes:
```
/usr/bin/cura5: line 5:  7483 Segmentation fault      LD_PRELOAD=/usr/lib/libstdc++.so.6 QT_QPA_PLATFORMTHEME=xdgdesktopportal QT_QPA_PLATAFORM=xdgdesktopportal ./UltiMaker-Cura $@
```

Jorengarenar, 2023-12-02:
Issue still present in 5.6.0

AdamScherlis, 2024-02-01:
Same issue in `aur/cura-bin 5.6.0-1` on EndeavourOS 2023.11.17 (KDE).

@geraldnilles trick worked for me too!

dj-foxxy, 2024-04-07:
On Arch Linux updated today `cura -platformtheme gtk3` worked.

krayon, 2024-08-03:
Worth noting that, at least for me, the `Save Project...` and `Export...` dialogs **_DO_** work.

erjoalgo, 2025-04-07:
I'm having the same issue in the latest UltiMaker-Cura-5.10.0-linux-X64.AppImage (5.10). Adding the -platformtheme gtk3 flag does work around the bug.