Ultimaker/Cura #16821: Slicer fails with apparently sound model.

sgparry, 2023-09-25:
### Cura Version

5.4.0

### Operating System

Ubuntu 22.04 (KDE Neon)

### Printer

Vertex Delta K8800

### Name abnormal settings

Fails with standard High profile.

### Describe model location

Managed to get a slice just once by combining several moves and rotations and fixes. Could not reproduce.

### Describe your model

Mesh check says it is watertight; tried mesh fix - no change

### Add your .zip here ⬇️

[VDK_Bohrschablone.3mf.gz](https://github.com/Ultimaker/Cura/files/12719705/VDK_Bohrschablone.3mf.gz)
mmm

GregValiant, 2023-09-26:
Thanks for the report.
The model has problems.  It looks like it was changed in CAD and the previous features weren't completely overwritten.

Here you can just make out an oval shape around what is now a round hole.  
[image]

Here it is sliced in PrusaSlicer.  As you can see...it's a mess.  (Kind of a cool looking face on that one boss though!).
[image]

This is a situation where it's really impossible to tell if the "no-slice" is caused by the model errors or a bug.  I was not able to repair the model with my usual tools.

So I'll go ahead and remove the bug label and close this as the model needs a re-work in CAD.  I hope you understand.

closed 2023-09-26 by GregValiant

sgparry, 2023-09-26:
Thanks. I wish people would post the original models on Thingiverse, not just the STLs - it's a lot easier to correct then.

GregValiant, 2023-09-26:
I always check for errors because they are often the problem.  Fixing errors is just an extension of that.
In this case it's just a really bad model.  I used formware.co/OnLineSTLRepair and MS 3D Builder and neither could repair this model.  MS 3D Builder is a problem because it's also my main tool for altering STL's.
If I really needed this piece I would convert it to a DXF file and draw a new model in AutoCAD using the DXF as a template.