Ultimaker/Cura #16822: When selecting Build Plate Adhesion Type to "SKIRT", Cura fails to gray out design if the skirt will print outside the printable area.

chimp33, 2023-09-25:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Creality Ender-3 v2

### Reproduction steps

1. Pick design such as bed levelling test and enlarge to near edge of printer area.
2. Select Build Plate Adhesion Type to "SKIRT" in settings
3. Cura fails to gray out design if the skirt will print outside the printable area.  When you go to slice, then print, the 3dp simply won't print that part of the skirt.

### Actual results

When you go to print it simply won't print that part of the skirt.

### Expected results

I would expect Cura to warn me by graying out the design and not allowing me to slice it.

### Add your .zip and screenshots here ⬇️

[CE3E3V2_PLA Fast Bed Level Test 0.45w_0.28h.3mf.zip](https://github.com/Ultimaker/Cura/files/12720555/CE3E3V2_PLA.Fast.Bed.Level.Test.0.45w_0.28h.3mf.zip)

GregValiant, 2023-09-26:
Thanks for the report.
That is actually a new feature introduced in 5.4.0.  Cura automatically adjusts the skirt so it no longer needs to be continuous if it goes off the bed.  That allows for larger prints to proceed that previously would not have sliced.
There can be a problem if you didn't leave any safety border around the build plate in Cura.  My Ender build plate is 235 x 235 but I leave it at 230 x 230 in Cura so there is a bit of a safety border.  Skirts that start heading off the build plate end at the edge of the safety border.
So this doesn't look like a bug to me.  I'll leave this open for a while so you can respond.

chimp33, 2023-09-27:
Thanks for the clarification then.

closed 2023-09-27 by chimp33