Ultimaker/Cura #16823: Slicing failed on t-rex arms

Coldarn, 2023-09-26:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Anycubic Kobra Go

### Name abnormal settings

The model sliced successfully on "Draft" 0.2mm layer height, but does not on "Normal" 0.15mm layer height

### Describe model location

Neither rotating nor translating the model allowed it to slice

### Describe your model

No, I've never sliced this model before

### Add your .zip here ⬇️

[CFFFP_TRex_Arms.zip](https://github.com/Ultimaker/Cura/files/12722021/CFFFP_TRex_Arms.zip)

GregValiant, 2023-09-26:
Thanks for the report and file.
There are three possible workarounds to fool this bug.  You tried two of them.

- Rotate the model
- Move the model
- Change the Initial Layer Height by a little bit.

I changed the Initial Layer Height to 0.23 and it sliced.

This looks like the normal slicing bug that 5.4.0 suffers from.  I'll mark this report as a duplicate and go ahead and close it as the Cura team has lots of examples of this behavior.  I hope you understand.

[image]

closed 2023-09-26 by GregValiant

Coldarn, 2023-09-26:
Thanks! I worked around the issue by changing the layer height.