Ultimaker/Cura #16830: remember the last opened (network) location when saving or opening a file

dbyy, 2023-09-26:
### Is your feature request related to a problem?

Currently when saving a gcode file (Save to Disk) the location dialog started always at my home location (Linux).
I than have to navigate to the just previously opened location (loading the STL file) to save the gcode.
Even if I made some changes in the settings (to tweak the slicing process) I have to start allover again to navigate around.

### Describe the solution you'd like

Cura should remember the last opened location and start there when saving a file or even open a new file.
I have all my 3D models on a network location (NAS) and I would like that Cura would remember the last used location to streamline the process of opening/saving files.

### Describe alternatives you've considered

I'm not aware of a suitable alternative

### Affected users and/or printers

Every user who uses Cura frequently will benefit from this improvement.

### Additional information & file uploads

PrusaSlicer (v 2.6.1) has this feature but I prefer using Cura v 5.4.
Thanks

alien8r, 2023-10-11:
I hoped that the recent 5.5.1-Beta might have fixed this failure to remember directory/file location: Nope.
There are many issues people have opened about this failure (over several releases) and it seems it is not being addressed.

dbyy, 2023-10-12:
Yeah, it looks lie no-one from the Cura team is reviewing the issues here on GitHub.
That's the only explanation I have for the fact that no-one has posted a comment to the reported issue, except 'alien8r'.