Ultimaker/Cura #16838: can't access my files on Cura

LabRoboticist, 2023-09-28:
### Cura Version

5.4.0 (moderen app image for linux)

### Operating System

Arch Linux (lts kernel)

### Printer

Creality Ender-3 / Ender-3 v2

### Reproduction steps

1 executed UltiMaker-Cura-5.4.0-linux-modern.AppImage
2 click on the file icon 
or 
2 go to file then click new project or open file(s) (the hot keys also cause the same issue)
Then Cura crashes with the same errors outputted on my terminal emulator.

### Actual results

PyInstaller/loader/pyimod02_importers.py:352: Deprecatio
 removal in Python 3.12; see the module's documentation
Cyclic dependency detected between "file:///tmp/.mount_U
tiMaCRuX6l/share/cura/resources/qml/Actions.qml"
file:///tmp/.mount_UltiMaCRuX6l/share/cura/resources/qml
tem: Binding loop detected for property "height"
file:///tmp/.mount_UltiMaCRuX6l/share/cura/resources/qml
tem: Binding loop detected for property "height"
/tmp/.mount_UltiMaCRuX6l/AppRun: line 25:  1970 Segmentation fault      (core dumped) $scriptdir/UltiMaker-Cura "$@"

Once I clicked on one of the file explorer options (from step 2) Cura crashes and goes away with these errors barfed on my screen. 

### Expected results

I hoped to access my files.

### Add your .zip and screenshots here ⬇️

I have been looking at forms and it seems to be a common problem, that people just ignore by using a graphical (draggable) file manager and dragging the files into Cura. This works for the time being but they aren't able to save their files. I like to keep my stuff minimal so I haven't tried it myself sense I just use my terminal emulator to access my files. I understand the pain of errors.

                                          
[cura.log](https://github.com/Ultimaker/Cura/files/12744717/cura.log)

LabRoboticist, 2023-09-28:
I don't know why Cura-5 is a hyperlink

phtalo, 2023-10-08:
I have the same issue, it clearly needs to be fixed but if you need to print something right now I suggest to try loading directly from terminal, e.g. `cura namefile1.stl namefile2.stl ...`. It should load them.

LabRoboticist, 2023-10-09:
Thank you for the help, but when I ran "UltiMaker-Cura-5.4.0-linux-modern.AppImage 3DBenchy_-_Multi-part_-_Single_-_Bridge_walls_-_3DBenchy.com.stl", in my terminal it did not open the file it just gave me an error message. It popped up as a little window at the bottom when Cura opened. 

"Unable to Open File"

"Failed to load /home/raspberry/3dBenchy_-_Multi-part_-
_Single_-_Bridge_walls_-_3DBenchy.com.stl. The file could
be corrupt, inaccessible or it did not contain any models."

I got the 3dBenchy file from Thingivers when I saw it used as a test print on YouTube. I got this file from "#3DBenchy - The jolly 3D printing torture-test by Creative" form "Creative Tools".