Ultimaker/Cura #16841: Slicing failed - unexpected error

john270, 2023-09-28:
### Cura Version

5.4

### Operating System

Windows 10

### Printer

Elegoo Neptune 4 pro

### Describe your model

Model fails to slice, regardless of settings or rotation

### Add your .zip here ⬇️

[EN4PRO_horse_center.zip](https://github.com/Ultimaker/Cura/files/12747427/EN4PRO_horse_center.zip)

GregValiant, 2023-09-28:
Thanks for the report.
This sliced for me on the first try so I'm unable to duplicate the problem.  This isn't the first time that has happened.  It might have something to do with the computer as this bug seems to be more prevalent on Windows machines.

I'll leave this for the Cura team.  Maybe they can use it for trouble-shooting.
You tried changing some settings, did you try small adjustements to the Initial Layer Height and/or Layer Height?

john270, 2023-09-29:
Thanks for the comments.  I did originally try changing layer height (which didn't help), but following your comment I've found a combination of layer height and initial layer height that's allowed it to compile (I set both to 0.15)

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic