Ultimaker/Cura #16844: Slicing Bug: radiused corner segments intermittently skipped

jonwienke, 2023-09-28:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Stock Ender 5 +

### Reproduction steps

When slicing STLs with radiused corners, the slicer occasionally skips part of the corner path, causing erroneous angled indentations in external corners, and extrusion lines floating in space on interior corners. The attached example shows the bug on exterior corners (circled).

### Actual results

The attached example shows the bug on exterior corners (circled). The walls in the top two visible layers should match the walls below.

### Expected results

Previous versions slice the project correctly (no erroneously indented/angled walls).

### Add your .zip and screenshots here ⬇️

[Cura Slicing Bug.zip](https://github.com/Ultimaker/Cura/files/12748988/Cura.Slicing.Bug.zip)
[image]

jonwienke, 2023-09-28:
Actually, you can see some interior corner issues in the screenshot as well. The opening on the upper left is supposed to have square corners, not chamfered at 45 degrees. And the inner wall between the opening and the left circled area should not be triangular.

GregValiant, 2023-09-28:
Thanks for the report.
This has been previously reported and I know I can duplicate the problem.  It does make it into the gcode so it isn't just a problem with the display.

I'll mark this as a duplicate of #15949 .  I'm going to leave it open for the Cura team to refer too.

jonwienke, 2023-10-06:
Actually, I spoke too soon. Turning **enable bridge settings** off reduces the probability of the corner bug occurring, but it **does NOT** make it go away completely.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic