Ultimaker/Cura #16846: Showing online resources does not work (e.g. Help->Show Online Documentation)

C2022H, 2023-09-29:
### Cura Version

5.4.0

### Operating System

openSUSE-LEAP 15.5

### Printer

Creality K1 Max

### Reproduction steps

Every function to show online resources will not work.
E.g. from the menu : Help->Show Online Documentation
The console from which I started the Cura Appimage shows for every action requiring online resources:
**kde-open5: /tmp/.mount_UltiMaoyoPs4/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib64/libicuuc.so.suse65.1)**
Looks like libicuuc.so.suse65.1 in the AppImage requires a different version of libstdc++

I am using UltiMaker-Cura-5.4.0-linux.AppImage, not the "modern" version of AppImage !!
The modern-version of the AppImage does not run on openSUSE-LEAP-15.5

### Actual results

Nothing happens except the error message in the console window

### Expected results

Online resource show up (in a browser or in an embedded window)

### Add your .zip and screenshots here ⬇️

[cura.log](https://github.com/Ultimaker/Cura/files/12762404/cura.log)

MariMakes, 2023-09-29:
Hi @C2022H 

Welcome to the UltiMaker Cura Github 🚀 
Thanks for your report 👍

I don't think your problem is linux specific. On Windows I'm being sent to an empty looking link aswell. 
[https://support.ultimaker.com/s/topic/0TO5b000000Q4utGAC/ultimaker-cura](https://support.ultimaker.com/s/topic/0TO5b000000Q4utGAC/ultimaker-cura)

I'll check with our support team where the online documentation is. 

However if you are looking for an insight in the different settings, I suggest you check out the Settings Guide Plug-in on the marketplace, it gives extra explanation with extra pictures for the different available settings. 

Thanks again!

C2022H, 2023-09-29:
The Help->Show Online Documentation was only an example, which you can easily reproduce.
In fact this problem occurs for extensions as well, if they provide some kind of online link.
You should find the reason for the error message 
**kde-open5: /tmp/.mount_UltiMaoyoPs4/libstdc++.so.6: version `GLIBCXX_3.4.30' not found (required by /usr/lib64/libicuuc.so.suse65.1)**, I assume this would fix the issue

C2022H, 2023-09-29:
By the way, your link
https://support.ultimaker.com/s/topic/0TO5b000000Q4utGAC/ultimaker-cura
leads to an empty page telling me "No articles found"

Beep6581, 2023-12-10:
This issue is about the actual click event in the AppImage not working.
For the problem of the "Show Online Documentation" URL leading the user to a page which says "No articles found" I created a new issue #17599.