Ultimaker/Cura #16850: interlocking

traiger, 2023-09-29:
### Cura Version

5.4

### Operating System

Windows 10

### Printer

S5

### Name abnormal settings

Interlocking

### Describe your model

simple keychain

### Add your .zip here ⬇️

[](url)
[UMS5_wbs key chain (1).zip](https://github.com/Ultimaker/Cura/files/12770884/UMS5_wbs.key.chain.1.zip)

GregValiant, 2023-09-30:
Thanks for the report and the file.
This wouldn't slice for me "as received".  I "selected all models" and moved the entire group to the left a little bit and towards the front a little bit.  After that it sliced.
[image]

This seems to be a duplicate of the popular and often reported slicing bug in 5.4.0.  Other workarounds are to rotate a model (not an option here) and/or change the Initial Layer Height by a little bit.

I'll mark this as a duplicate.  Since it's a UM printer I'll leave this open for the Cura team.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic