Ultimaker/Cura #16852: Cura 5.4.0 Model Slice Fail

itzderp, 2023-09-30:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Elegoo Neptune 3 Pro

### Describe model location

It slices if I rotate the model 90 degrees

### Describe your model

It slices when I rotate it 90 degrees and I've tried the meshtools plugin as well

### Add your .zip here ⬇️

[EN3PRO_helm_pt03.zip](https://github.com/Ultimaker/Cura/files/12773990/EN3PRO_helm_pt03.zip)

GregValiant, 2023-09-30:
Thanks for the report.
The model is watertight but as you can see, one of the surfaces above the "earhole" is either extraneous, or is inverted.  Cura shows it in red indicating it requires support, but it's facing upward and would not need support.  It's the sort of model error that can confuse the slicer.
Here you can see that the surface is sort of floating in space.  I wouldn't trust this to slice correctly.
[image]

There are parts of this model that will require support.  You current settings don't allow for the support to develop.

I'll leave this open for the Cura team.  Both the model and some settings seem to require adjustment.  That makes it tough to determine if the problem is caused by a bug or by something else.

HellAholic, 2024-04-03:
We made some improvements to prevent slicing fails in Cura 5.7.0 available to download now. I was able to slice your model successfully in 5.7.0. Could you test if that's the case for you as well? Feel free to reopen the issue if it is not resolved.
You can download it here: https://github.com/Ultimaker/Cura/releases/tag/5.7.0

closed 2024-04-03 by HellAholic