Ultimaker/Cura #16854: Rose

JaneBelousova, 2023-09-30:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Flying bear ghost 6

### Describe your model

Can't slice because of unexpected error

### Add your .zip here ⬇️

[CFFFP_rose.zip](https://github.com/Ultimaker/Cura/files/12775373/CFFFP_rose.zip)

GregValiant, 2023-09-30:
Thanks for the report.
I had to rotate it 90° CCW and then it sliced.

I'll mark this as a duplicate as this bug has been reported numerous times.  I'll go ahead and close it as the Cura team has plenty of examples of this behavior.  I hope you understand.

closed 2023-09-30 by GregValiant

JaneBelousova, 2023-10-01:
Slice become successfull after rotating around Z axis. But it depends on the angle: in some cases the model couldn't be sliced.

JaneBelousova, 2023-10-01:
> Thanks for the report. I had to rotate it 90° CCW and then it sliced.
> 
> I'll mark this as a duplicate as this bug has been reported numerous times. I'll go ahead and close it as the Cura team has plenty of examples of this behavior. I hope you understand. [image]

Thanks for reply and help!