Ultimaker/Cura #16857: Sovol printer definition overrides material default print temperature setting

mstrent, 2023-10-01:
### Cura Version

5.4.0

### Operating System

Any

### Printer

Sovol SV06

### Reproduction steps

Install Cura and configure for Sovol SV06. Slice anything and print it.

### Actual results

 Observe print temperature will ALWAYS be 195*, regardless of selected material print temperature.

### Expected results

Print temperature should inherit material print temperature setting.

### Add your .zip and screenshots here ⬇️

This is caused by the Sovol printer definition statically defines and overrides the temperature.

mstrent, 2023-10-01:
Easy fix: PR #16856 should resolve. Have tested it by altering the file on a Cura 5.4.0 install on my PC.

GregValiant, 2023-10-01:
Another issue with the Sovol definition files is addressed in #16107 .

mstrent, 2023-10-01:
Closing as duplicate of #16100, my bad. Thanks @GregValiant!

closed 2023-10-01 by mstrent