Ultimaker/Cura #16859: Slice Fail on Ender 3 Pro for 214 x 214 x 8mm print

star-dust-1, 2023-10-01:
### Cura Version

5.4.0

### Operating System

Windows 11

### Printer

Ender 3 Pro

### Name abnormal settings

10 percent infill

### Describe model location

Slices if scaled down

### Describe your model

A square frame with rectangular cutouts 

### Add your .zip here ⬇️

[CE3PRO_Ender3 Magnet Frame_v0.zip](https://github.com/Ultimaker/Cura/files/12776514/CE3PRO_Ender3.Magnet.Frame_v0.zip)

It allowed me to slice when I downsized the object to 208 x 208 mm but theoretically the print area should be 220x220 mm

GregValiant, 2023-10-01:
,Thanks for the report.
With the model at 214 x 214 you have to add the Bed Adhesion settings.
skirt_loops_width + skirt_distance + model width + skirt_distance + skirt_loops_width = Print Footprint X 
skirt_width + skirt_distance + model depth + skirt_distance + skirt_width = Print Footprint Y

If you have a line width of 0.4, and you are printing 3 loops in the skirt and the skirt distance is 10mm, then the actual space the print takes up on the bed (the "Footprint") would be 236.4 and so it would be over the "220" listed in the machine settings.

The skirt/brim behavior was changed in 5.4(?).  Now, if any little bit of skirt/brim will fit on the build plate then then model will slice.
Here is your model with a skirt (3) loops wide and at a distance of 5mm from the model.
[image]

Only that little bit of skirt makes it onto the build surface (at each corner) but that is enough to allow this to slice.  If the "skirt distance" was 6mm this would refuse to slice.  The total "footprint" would be larger than the 220 width/220depth of the build surface.

I'll leave this open for now, but it sounds like a misunderstanding and not a bug.
Let us know if changing the Bed Adhesion setting allows this to slice.

star-dust-1, 2023-10-01:
Thanks for the speedy reply! I wasn't aware of the special considerations for a part that was so close to the max build plate area

GregValiant, 2023-10-01:
One more thing...Most Enders have purge lines (within their startup gcode) that run up the left side.  You need to grab them and pull them off before the nozzle comes back around.

I'll go ahead and close this as it's expected behavior.  If you want to use the whole bed (or most of it) you need to set the Bed Adhesion to None, and be ready to pull the purge lines off.

closed 2023-10-01 by GregValiant